atomic14/esp32-s3-pinouts #2: Are there pins that should be isolated (if not used) during sleep, to minimize current consumption?

peteDDD, 2023-11-24:
Again, really great, and needed reference.

On the original ESP32, GPIO_12 was normally isolated during sleep to save power.

`  // Isolate GPIO12 pin from external circuits. This is needed for modules
  // which have an external pull-up resistor on GPIO12 (such as ESP32-WROVER)
  // to minimize current consumption.
  rtc_gpio_isolate(GPIO_NUM_12);`

Have you found any pins that should be similarly isolated on the ESP32-S3?

cgreening, 2023-11-24:
Thanks!

As far as I can see, there's nothing mentioned in the data sheet

peteDDD, 2023-11-24:
I wonder if such was "mentioned" in the ESP32 data sheet or was this just something folks found along the way.

cgreening, 2023-11-24:
Hmm there's this: https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32s3/api-reference/system/sleep_modes.html#configuring-ios

Maybe it's referring to some of the strapping pins that have default pull up/down