atomic14/esp32-s3-pinouts #17: JTAG pins (GPIO 39-42) default config?

SmittyHalibut, 2024-05-01:
Documentation I'm reading: 
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf#subsubsection.2.6.4

[image]

The way I read section 2.6.4, Table 2-13, if we assume that all the eFuses are 0, then GPIO3 is ignored on start-up and JTAG is on USB, NOT on the JTAG pins.  This means that GPIO39-42 should default to normal GPIO use.  But this (wonderful!) reference says otherwise.

Am I missing something here?  Are eFuses not all 0 by default? Am I looking at the wrong document?  Is Espressif's documentation wrong?

Mr-HaleYa, 2024-05-04:
By default those three E-fuse bits are 0, you must bun them yourself and then it can never be undone.
GPIO3 is safe to use as long as eFuse3 is 0
GPIO39-42 default as regular GPIOS unless the fuses are burned as the table shows

SmittyHalibut, 2024-05-04:
Ok, that's how I read it too. But README.md disagrees. It says the default for those pins is to act as JTAG.

cgreening, 2024-05-05:
Sounds like it needs a bit of a tweak to make it clear the pins are ok unless you've messed with fuses - would you like to make a PR?

SmittyHalibut, 2024-05-06:
"Good idea!  I accept Pull Requests!"  I like the cut of your jib, sir. :-)

Yeah, I will.  Stand by, I've got some other time pressing tasks today.  Are you able to assign this ticket to me, so it shows up in my list of things to do and I don't forget about it?

SmittyHalibut, 2024-05-23:
Bumping to keep this alive.  This is on my to-do list, but I've been super busy with conventions for work.  Don't close it yet, sorry.

SmittyHalibut, 2024-06-06:
PR Submitted.  https://github.com/atomic14/esp32-s3-pinouts/pull/21

cgreening, 2024-06-06:
Merged! Thanks so much for your contribution.