atomic14/esp32-s3-pinouts #18: GPIO47/48 voltage

higaski, 2024-05-08:
I've a suggestion regarding GPIO47/48. On some S3 boards (e.g. the wroom-2) those pins only have 1.8V working voltage. This is mentioned on page 12 in the [Wroom-2 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf).

Thank you for this repository, great overview of the S3 pinout!

robb-brown, 2024-05-22:
My thanks as well! 

I just came to add this as I happened to come across that datasheet note. Apparently those pins are probably not 3.3V tolerant, so this is a pretty big item on affected versions.

cgreening, 2024-06-06:
I'll take a look - sounds like it could cause some horrible problems!